StuartsProjects/SX12XX-LoRa #69: Transparent Mode for SX128X with stm32f103 controller

karthikgn, 2023-09-07:
Hi.,

I am trying Transparent mode in sx128X with stm32f103 controller. I am able to transmit from one side and the other side i am receiving the data, but the data has to send from both side not working. Help me to resolve this.

Thank You.,
Regards
Karthik

StuartsProjects, 2023-09-07:
You will need to explain exactly what you mean by 'Transparent Mode'. 

Which of the library examples use this 'Transparent Mode' ?

karthikgn, 2023-09-08:
Transparent Mode means full duplex mode, By using TransmitIRQ and ReceiveIRQ in main code its not working to achieve full duplex operation.

StuartsProjects, 2023-09-08:
The SX128X (and SX126x, SX127X) are half duplex devices, at any one time they can only be in receive or transmit mode.

karthikgn, 2023-09-08:
SX128X used in ebyte module similar operation has to perform using your library  by using stm32f103 controller, help me to do with your library.

StuartsProjects, 2023-09-08:
So your trying to replicate the functionality of the eByte E28 modules, why exactly, the eByte modules are low cost ?

karthikgn, 2023-09-08:
Need only the functionality of the ebyte

StuartsProjects, 2023-09-08:
Well eByte will be using buffers and timeouts to achieve the impression of full duplex, but that is not actually possible with LoRa devices, they can only be in receive OR transmit mode. 

Thus for your application you will need to switch constantly between receive and transmit modes, and there are plenty of examples showing how this is done in the context of send and acknowledge processes in the \reliable examples folder. There is a lot more to your appplciation than setting appropriate IRQs. 

I provide support for issues\faults with the library and the provided examples, but do not provide free support for custom applications.

StuartsProjects, 2024-01-07:
The SPI LoRa modules are not capableof full duplex.

closed 2024-01-07 by StuartsProjects

karthikgn, 2024-01-08:
if i use 2 module this can be achieved?