StuartsProjects/SX12XX-LoRa #71: "Send array attempt 1 Start array transfer for /pic225.jpg"  after that serial note device Freeze.

rtarta, 2023-09-22:
Hi Dear @StuartsProjects  first of all i want to thank you for your project. 
SX127X series transmeter esp32cam (project example no: 240)  after a while device freeze.  I think that ARsendArray at method. 
Do you have any recommend for that case ?:

```
Send with LoRa /pic225.jpg
Send array attempt 1
Start array transfer for /pic225.jpg
```

StuartsProjects, 2023-09-23:
So its taken and transferred 224 pictures and then stops. 

If you restart it does it stop at other numbers of transfers less than 225 or more than 225. 

What are you transfering the pictures too ?

rtarta, 2023-09-23:
@StuartsProjects  Yes it takes  after awhile it stops. 
when restart it continue to sending  picture number may change.  Picture size max 60 KB each 2 minute .
but let me test more time.

rtarta, 2023-09-30:
@StuartsProjects Hi, i figured out that image size more than 60KB that freeze. File size affect the start transmission.  Do you have any recommendation about that case ?

rtarta, 2023-10-05:
@StuartsProjects  i got the problem with SX1278 in  SX127XLT::CRCCCITT  size overload is small for getting data over 64KB.  I have change uint16_t to uint32_t libraryCRC and index variable it worked but now stops on 128 KB  image.  what if i want to change them to uint64_t ?.

StuartsProjects, 2023-10-05:
Changing the index to uint32_t allows the function to read an array that is 4,294,976,295 bytes long way more memory that is on the ESP32. 

Changing the index to uint32_t work for me, I have had the 240 sketch running for some hours, saving files that are 170.000bytes +.

Where does you program stop ?

What changes to the settings for the camera did you make ?

rtarta, 2023-10-05:
Device stop insede ARtransferIRQ.h: => ARstartArrayTransfer method. (Line 315)
Below  changed the code  
`
//In File ARtransferIRQ.h:
bool ARsendArray(uint8_t *ptrarray, uint32_t arraylength, char *filename, uint8_t namelength); 

//In File SX127XLT.cpp 
uint32_t SX127XLT::CRCCCITT(uint8_t *buffer, uint32_t size, uint32_t startvalue)
{
#ifdef SX127XDEBUG1
  Serial.println(F("CRCCCITT() "));
#endif

  uint32_t index, libraryCRC;
  uint8_t j;

  libraryCRC = startvalue;                                  //start value for CRC16

  for (index = 0; index < size; index++)
  {
    libraryCRC ^= (((uint32_t)buffer[index]) << 8);
    for (j = 0; j < 8; j++)
    {
      if (libraryCRC & 0x8000)
        libraryCRC = (libraryCRC << 1) ^ 0x1021;
      else
        libraryCRC <<= 1;
    }
  }

  return libraryCRC;
}`

StuartsProjects, 2023-10-06:
```
uint32_t index, libraryCRC;
uint8_t j;
```

Thats is not correct. 

Its index only that needed changing to uint32_t, libraryCRC needs to stay at uint16_t, this is how the code should look;

```
uint32_t index;
uint16_t libraryCRC;
uint8_t j;
```

Can you restrict changes to the 240 Sketch to only those affecting the image resolution or size. I have not tested items such as changing the grab mode, and neither do I support them. The ESP32CAM part of the code is from the RandowNerdTutorials site, if you want support for ESP32CAM specific issues then maybe ask them.

rtarta, 2023-10-06:
Thank you for you details, i will change as you recommend. Camera config only for taking the best frame in tutorial camera RandowNerdTutorials, camera takes first frame and that is Green tint so image is little bir green.  For getting best image frame i changed. 🙏
config.grab_mode = CAMERA_GRAB_LATEST;
config.fb_count = 2;

rtarta, 2023-10-06:
@StuartsProjects 
Below i got error but not freeze the device as updated the code above comment.  Now getting ERROR destination array and local array CRCs do not match. File size 168228 bytes. Can you update me if possible what is the reason. thanks.

18:50:23.025 -> Send with LoRa /img164_2769DBBAFE5C.jpg
18:50:23.025 -> Send array attempt 1
18:50:23.025 -> Start array transfer for /img164_2769DBBAFE5C.jpg
18:50:23.058 -> Send open remote file request
18:50:23.091 -> Send attempt 1
18:50:23.123 -> Array write started OK on remote
18:50:23.156 -> Source array length 168228 bytes
18:50:23.156 -> Source array CRC is 0xEDCD
18:50:23.156 -> Segment Size 245
18:50:23.156 -> Number segments 687
18:50:23.156 -> Last segment size 158
18:50:23.156 -> 0
18:50:23.255 -> 1
18:50:23.387 -> 2
***
***
18:51:40.714 -> 686
18:51:40.817 -> Send end array write
18:51:40.916 -> Array write ended OK on remote
18:51:40.916 -> Acknowledged remote destination file length 168228
18:51:40.916 -> Destination array and local array lengths match
18:51:40.916 -> Acknowledged destination array CRC 0x9210
18:51:40.916 -> *********************************************
18:51:40.916 -> ERROR destination array and local array CRCs do not match
18:51:40.916 -> Restarting transfer
18:51:40.916 -> *********************************************
18:51:40.948 -> Send array attempt 2
18:51:40.948 -> Start array transfer for /img164_2769DBBAFE5C.jpg
18:51:41.014 -> Send open remote file request
18:51:41.014 -> Send attempt 1
18:51:41.522 -> Send open remote file request
18:51:41.556 -> Send attempt 2
18:51:42.027 -> Send open remote file request
18:51:42.060 -> Send attempt 3
18:51:42.563 -> Send open remote file request
18:51:42.597 -> Send attempt 4
18:51:43.136 -> Send open remote file request
18:51:43.136 -> Send attempt 5
18:51:43.610 -> *************************
18:51:43.610 -> ERROR writing to remote array
18:51:43.610 -> Restarting transfer
18:51:43.610 -> *************************
18:51:44.592 -> /img164_2769DBBAFE5C.jpg Send picture failed
18:51:44.626 -> ********************************
18:51:44.626 -> ERROR - Take picture send failed

rtarta, 2023-10-06:
Hi, i have started from examples from library and did as you said.  I got below result.  My lora device is LoRa-02 SX1278 433 MHZ.

21:52:19.848 -> 401
21:52:19.980 -> 402
21:52:20.078 -> 403
21:52:20.176 -> 404
21:52:20.207 -> 405
21:52:20.240 -> /pic3.jpg end array write request
21:52:20.240 -> ARDTDestinationArrayLength 99243
21:52:20.272 -> Destination arrayCRC 0x5223
21:52:20.272 -> Array write ended
21:52:20.272 -> Transfer time 45903mS
21:52:20.272 -> Destination array length 99243 bytes
21:52:20.272 -> Source array length 99243 bytes
21:52:20.272 -> Array lengths match
21:52:20.272 -> Destination array CRC is 0x5223
21:52:20.272 -> Source array CRC is 0xE35B
21:52:20.272 -> ERROR - array CRCs do not match
21:52:20.304 -> Returned picture length 99243
21:52:20.370 -> SD Card OK
21:52:20.370 -> /pic3.jpg Save picture to SD card
21:52:20.927 -> /pic3.jpg Saved to SD
21:52:20.927 -> 
21:52:20.927 -> LoRa file transfer receiver ready

rtarta, 2023-10-07:
Dear @StuartsProjects  in 241 Receiver 
const uint32_t ARDTarraysize = 0x20000;       was  like that even i tried to send it does not get the file so i change the  value to  0x40000 now its working. Thanks for your supports.

StuartsProjects, 2023-10-11:
I have tested revised examples and library file ARtransferIRQ.h which allow for much larger images to be sent. I have tested for SX127x but before publishing the update I need to check the changes against SX126x and SX128x.

rtarta, 2023-10-12:
Thank you, i am closing that case related with SX127X. 
thank you for your supports.

closed 2023-10-12 by rtarta