StuartsProjects/SX12XX-LoRa #72: No Receive

ArshaanTech, 2023-10-11:
I am trying example code for IRQ. 
But I do not receive anything at the receiver end.
This is the Tx window of Arduino LoRa:
[image]

StuartsProjects, 2023-10-11:
The library example code does work.

However, I cannot possibly help unless you provide a lot more information, such as detail of which programs you are using for transmit and receive, the LoRa modules you are using, the Arduinos you are using, how far apart transmitter and receiver are,  and how you have it all connected.

StuartsProjects, 2023-10-11:
Those Ra01SH modules have TX and RX Enable pins, none of the standard examples support that setup although examples 103_LoRa_Transmitter_Detailed_Setup and 104_LoRa_Receiver_Detailed_Setup do and may work 

SX126X modules with TX and RX Enable pins have not been tested with the library, as is explained in the library documentation;

_SX126x modules with RX or TX enable pins
Some SX126x modules may have RX or TX enable pins, Ebyte E22 modules for instance. Whilst there is code in the SX126X library that was copied across from the SX128X library there were no SX126X devices using RX and TX switching available at the time this library was published, so these functions are currently not supported by the library as they have not been tested. If a user has SX126X modules that they want tested with the library, then donations of modules will be accepted._

ArshaanTech, 2023-10-11:
Okay... Thank you. 
I will try 103_LoRa_Transmitter_Detailed_Setup and 104_LoRa_Receiver_Detailed_Setup. if you have any other way of Tx and Rx pins do let me know.

ArshaanTech, 2023-10-11:
No luck... same issue is persisting...

StuartsProjects, 2023-10-12:
And which Arduinos are you using ?

ArshaanTech, 2023-10-12:
I am using Arduino UNO standard ATMega 328.

StuartsProjects, 2023-10-12:
The Arduino UNO is a 5V logic level device, you should not connect it directly to a 3.3V logic level LoRa device such as the RA01SH.

ArshaanTech, 2023-10-13:
But it is working fine with the Radiolib Library... That somehow resolved the problem... also I think my module has extra capacitors stuff coz I am powering it with directly 5v and it's transmitting and receiving data. The Radiolib library is working fine.

StuartsProjects, 2023-10-13:
Well as I said I have not tested those particular modules and apart from potential TX and RX enable issues there are other settings that can vary between modules. If someone were to donate a set of modules I would test them.

However, if you do not have the modules connected appropriately, who knows what the problem could be.

ArshaanTech, 2023-10-13:
okay got it... I am sorry but I don't think i'll be able to donate my modules as I am in India and it would also be high cost for transportation...
Anyways Thank you so much! 
I am using the RadioLib library for now which is working fine

closed 2023-10-13 by ArshaanTech

miszaklajent, 2024-05-18:
I think I have a similar problem. I'm running [E22-900MM22S SX1262](https://www.cdebyte.com/products/E22-900MM22S/2#Pin) with rpi pico. I run the **2_Register_Test** and it spits in serial this: 
[image]
[image]
for node one, and 
[image]
[image]
for the second node, (the setups are exactly the same)
So I would assume that communication is working? (between lora module and controller)
I've tried the **103_LoRa_Transmitter_Detailed_Setup** and **104_LoRa_Receiver_Detailed_Setup** with not much luck, basically, it looks like it doesn't ever go past the `if (LT.transmit(buff, TXPacketL, 10000, TXpower, WAIT_TX))` (I used serial print stamps to analyze that), so it looks like it doesn't ever actually send the data. When I checked pins RXEN and TXEN, the first one was pulled low and the second high, so looks like that part works. I'll try to look around it a little more, maybe I'll find something helpful.

StuartsProjects, 2024-05-18:
Please note this issue was closed. 

Subsequent to closure the SX126x library was tested against an E22-900M22S that requires RX and TX enable pins to be driven, active high, and the library operated correctly, no fault found.

If you have a seperate issue then please rate a new one.