StuartsProjects/SX12XX-LoRa #74: Problem sending and receiving data after reboot

trlink, 2023-12-04:
Hi all,

I have a problem with the library sending and receiving data after a reboot (can reproduce it with ESPs and T-Beams):

After first startup, the modem is sending an receiving as it should, when I update the ESP (or the ESP crashes), sending and receiving doesn't work. When this problem occurs, I can see, that the transmit() function will take much less time...

Does anybody have an Idea?

Kr

Christian

StuartsProjects, 2023-12-05:
If the SX127X is left in some indetermiante state because of a crash in between operations, then the library has no possible way of detecting that the sketch has 'crashed' or that IO pin functions might have changed.

You need to find a way to detect the 'crash' and re-intialise the SX127X registers and IO pins to a known state, as if from power on or reset, sounds like an ESP32 specific issue.

trlink, 2023-12-07:
Hi Stuart,

I already call resetDevice() after startup, how can I achieve a re-initialization?

StuartsProjects, 2023-12-07:
Follow the initialisation process thats in the setup() function of the library examples. 

The full detail of the  initialisation process is in sketch '103_Lora_Transmitter_Detailed_Setup' and described in detail in the library Readme file.

trlink, 2023-12-09:
Problem exist between keyboard and chair... Was my fault, my init was wrong, called setupLoRa() before other init functions to setup the modem... You can close this issue...

closed 2023-12-09 by trlink