StuartsProjects/SX12XX-LoRa #75: Support for ESP Watchdog

trlink, 2023-12-09:
Hi Stuart,

when using the library in a multi threaded application, where you have a thread for receiving and transmitting data, I ran into the problem, that when I wait for a packet, the watchdog gets triggered. If the timeout for the receive() function is to low, I cant receive the complete packet, a full packet will always end up in a WDT_RST...

I already changed it by myself, but I would like to ask, if you can support the watchdog by simply adding a wdt reset tom the loop:

`  //Wait for DIO1 to go high
  while (!digitalRead(_RXDonePin))
  {
    delay(10);
    esp_task_wdt_reset();
  };   `

StuartsProjects, 2023-12-09:
The SX12XX library deliberatly does not include functions that are specific to particular microcontrollers. 

With so many microcontrollers available these days, and more appearing on a regular basic, introducing 'features' for particular microcontrollers would introduce significant support issues.

StuartsProjects, 2024-01-07:
There are no plans to modify the library to include support for individual Arduinos

closed 2024-01-07 by StuartsProjects