StuartsProjects/SX12XX-LoRa #78: Throughput at SF5

SharathChandan, 2024-01-11:
Is it possible to get a throughput of 60 kbps consistently in SF5 and 500kHz configuration of LoRa? I am using an RFM95W LoRa module connected to an ESP32. I encounter a send error while trying to transmit data.

StuartsProjects, 2024-01-11:
> I am using an RFM95W LoRa module

The RFM95 uses an SX127X LoRa module which does not support SF5.

SharathChandan, 2024-01-11:
Thanks for clarifying. If I use the sx1268 modules instead of rfm95w, can it help in achieving close to 60 kbps?

StuartsProjects, 2024-01-11:
> Is it possible to get a throughput of 60 kbps consistently in SF5 and 500kHz configuration of LoRa? I am using an RFM95W LoRa module connected to an ESP32. I encounter a send error while trying to transmit data.

The issue you have raised is that when configuring an RFM95 (SX127x) for SF5 you were getting send errors. 

The library should not allow you to compile a sketch for a SX127X at SF5, so there would be no program to upload and run.

SharathChandan, 2024-01-12:
> Thanks for clarifying. If I use the sx1268 modules instead of rfm95w, can it help in achieving close to 60 kbps? Thanks for creating such a comprehensive library.

I'm facing error in other spreading factors too? is it related to hardware issue? Can sx1268 serve my purpose of using SF5 and getting close to 60 kbps?

StuartsProjects, 2024-01-12:
> I'm facing error in other spreading factors too? is it related to hardware issue? Can sx1268 serve my purpose of using SF5 and getting close to 60 kbps?

Can you provide details of which example in the library is producing 'errors' and what those errors are ?

60kps is the maximum air rate of a packet on SX126x. How close you can get to that in a real application would depend on the application and how long the application takes to fetch the data, process it and load it into the LoRa device.

SharathChandan, 2024-01-17:
I managed to resolve the send errors. But the receiver doesn't print the sent payload. While using the 42 throughput test sketch, I face NoAcK error.

StuartsProjects, 2024-01-17:
Can you provide some more information ?

First I need to know which LoRa module you are using and which microcontroller ?

None of the examples as provided to my knowledge would need 'errors resolved' to make them work, so some details here of what you did are needed. 

The sketches as published should work, if your making changes to them you need to say if the original sketches work for you and if they stopped working when you make changes.

SharathChandan, 2024-01-20:
I am using an ESP32S microcontroller along with an RFM95W module. I wanted to test the max throughput of my SX1276 at Spreading Factor 6 and BW 500 KHz. 

#define SCK 18                                  //SCK on SPI3
#define MISO 19                                 //MISO on SPI3 
#define MOSI 23                                 //MOSI on SPI3 

#define NSS 5                                   //select pin on LoRa device
#define NRESET 27                               //reset pin on LoRa device
#define RFBUSY -1                               //busy line
#define DIO1 35                                 //DIO1 pin on LoRa device, used for RX and TX done 
#define LED1 2                                  //on board LED, high for on
#define DIO0 2 

#define LORA_DEVICE DEVICE_SX1278               //we need to define the device we are using

//*******  Setup LoRa Parameters Here ! ***************

//LoRa Modem Parameters
const uint32_t Frequency = 868000000;           //frequency of transmissions in hertz
const uint32_t Offset = 0;                      //offset frequency for calibration purposes

const uint8_t Bandwidth = LORA_BW_500;          //LoRa bandwidth
const uint8_t SpreadingFactor = LORA_SF6;       //LoRa spreading factor
const uint8_t CodeRate = LORA_CR_4_5;           //LoRa coding rate
const uint8_t Optimisation = LDRO_AUTO;         //low data rate optimisation setting, normally set to auto

const int8_t TXpower = 14;                      //LoRa transmit power in dBm

//*******  Setup packet parameters Here ! ***************
const uint8_t numberPackets = 50;               //number of packets to send in transmit loop
const uint8_t TXPacketL = 240;                   //length of packet to send  
const bool waitforACK = true;                   //set to true to have transmit wait for ack before continuing

SharathChandan, 2024-01-20:
My receiver code:

/*******************************************************************************************************
  Programs for Arduino - Copyright of the author Stuart Robinson - 25/03/20

  This program is supplied as is, it is up to the user of the program to decide if the program is
  suitable for the intended purpose and free from errors.
*******************************************************************************************************/

//*******  Setup hardware pin definitions here ! ***************

//These are the pin definitions for one of my own boards, the Easy Pro Mini, 
//be sure to change the definitions to match your own setup. Some pins such as DIO1,
//DIO2, BUZZER may not be in used by this sketch so they do not need to be
//connected and should be set to -1.

#define NSS 5                                  //select pin on LoRa device
#define NRESET 27                                //reset pin on LoRa device
#define LED1 8                                  //on board LED, high for on
#define DIO0 2                                  //DIO0 pin on LoRa device, used for RX and TX done 

#define LORA_DEVICE DEVICE_SX1278               //we need to define the device we are using

//*******  Setup LoRa Parameters Here ! ***************

//LoRa Modem Parameters
const uint32_t Frequency = 868000000;           //frequency of transmissions in hertz
const uint32_t Offset = 0;                      //offset frequency for calibration purposes

const uint8_t Bandwidth = LORA_BW_500;          //LoRa bandwidth
const uint8_t SpreadingFactor = LORA_SF6;       //LoRa spreading factor
const uint8_t CodeRate = LORA_CR_4_5;           //LoRa coding rate
const uint8_t Optimisation = LDRO_AUTO;         //low data rate optimisation setting, normally set to auto

const int8_t TXpower = 14;                      //LoRa transmit power in dBm

StuartsProjects, 2024-01-20:
You have not said which of the library examples you are using !!!!

All you have provided so far is the settings you are using for an unknown example ?

The pin definitions look a bit odd also, have you confirmed that the basic examples '3_LoRa_Transmitter' and '4_LoRa_Receiver' are working as standard ?

closed 2024-02-01 by SharathChandan

SharathChandan, 2024-02-03:
I found the issue to be in the pin connections, sir. Can you help me understand the RFBUSY pin, is it a physical pin on the LoRa module that needs to be connected to the ESP32? I have included the hardware used below.
[image]

StuartsProjects, 2024-02-03:
I have asked that you specify which type of LoRa modules you are using (and just pictures do not help) and which library examples you are using that do not work. 

So until you do that, I am sorry, I cannot help you.

SharathChandan, 2024-02-04:
> I have asked that you specify which type of LoRa modules you are using (and just pictures do not help) and which library examples you are using that do not work. 
> 
> So until you do that, I am sorry, I cannot help you. 

Sure sir. I'm using SX127x LoRa module. The library example used is in SX127x examples - ESP32 - Basics - Receiver and the Throughput transmitter test sketch. The receiver stops receiving after a single packet. The throughput sketch shows noAck error.

StuartsProjects, 2024-02-04:
> Sure sir. I'm using SX127x LoRa module. The library example used is in SX127x examples - ESP32 - Basics - Receiver and the Throughput transmitter test sketch. The receiver stops receiving after a single packet.

If you have set the spreading factor to SF6 then that is expected. On the SX127X SF6 requires that all packets are sent and received in fixed\known length implicit mode, see '40_LoRa_Transmitter_ImplicitPacket' and '41_LoRa_Receiver_ImplicitPacket' for examples of this mode. 

Very few of the SX127X examples will work in fixed\known length implicit mode, one of the limitations of that device, SF7 is the lowest spreading factor normally used. 

For high throughput applications the SX126X is a far bettwr device choice, there is operation down to SF5 in normal varialble length packet mode. 

SX127X devices do not have a RFBUSY pin.

SharathChandan, 2024-02-04:
Thanks for clarifying sir. Will get the SX1262 module instead and test the sketches.

SharathChandan, 2024-02-04:
Thanks for clarifying, sir. I will get the SX1262 module instead and test the sketches.

StuartsProjects, 2024-02-06:
Closd sinc no issues found with the library or example sketches. 

SF6 on an SX127X requires  specific code and example sketches are provided in the library.

If switching to SX126X modules do check if the pin conenctions are correct and supported by the SX126x part of the library.