StuartsProjects/SX12XX-LoRa #80: 236_SDfile_Transfer_TransmitterIRQ - Guru Meditation Error

TomsGitBits, 2024-03-02:
**Disclaimer**: There is a good chance this is not an issue with the library, but rather user error. I am a SWE by trade but entirely new to the hardware/electrical engineering/radio side of things. Apologies in advance for any ignorance on display. 

I purchased a pair of [LilyGo T3-S3 dev boards](https://www.tindie.com/products/lilygo/lilygo-t3-s3-v10-esp32-s3-lora-sx1280-24g/) which come outfitted with an ESP32-S3 chip, an SX1280 2.4g LoRa module, a micro-SD module, and a USB-C port. This board seemed like the easiest starting point given my inexperience. Upfront question: Does this library support the "S3" variant of the ESP32 chip?

The manufacturer of this board provides some basic LoRa Transmitter/Receiver [examples](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt/SX128x_Transmit_Interrupt.ino) that I have successfully been able to get working, but so far have been unsuccessful in running any of the StuartsProjects/SX12XX-LoRa examples. In particular, I'm working with the 236_SDfile_Transfer_TransmitterIRQ and 237_SDfile_Transfer_ReceiverIRQ programs.

The below pin definitions were supplied by the manufacturer of my board and are working properly on the examples they provided:
`#define I2C_SDA                     18
#define I2C_SCL                     17
#define OLED_RST                    UNUSE_PIN

#define RADIO_SCLK_PIN              5
#define RADIO_MISO_PIN              3
#define RADIO_MOSI_PIN              6
#define RADIO_CS_PIN                7
#define RADIO_DIO1_PIN              9
#define RADIO_DIO2_PIN              33
#define RADIO_DIO3_PIN              34
#define RADIO_RST_PIN               8
#define RADIO_BUSY_PIN              36

#define RADIO_RX_PIN                21
#define RADIO_TX_PIN                10

#define SDCARD_MOSI                 11
#define SDCARD_MISO                 2
#define SDCARD_SCLK                 14
#define SDCARD_CS                   13

#define BOARD_LED                   37
#define LED_ON                      HIGH

#define BAT_ADC_PIN                 1
#define BUTTON_PIN                  0

#define HAS_SDCARD
#define HAS_DISPLAY`

Mapping those pin definitions over to the  236_SDfile_Transfer_TransmitterIRQ settings gives me the below configuration:
`#define NSS 7                                  //select pin on LoRa device
#define NRESET 8                                //reset pin on LoRa device
#define RFBUSY 36                                //RFBUSY pin on LoRa device
#define LED1 37                                  //LED used to indicate transmission
#define SDCS 13
#define Monitorport Serial                      //Port where serial prints go

#define LORA_DEVICE DEVICE_SX1280               //this is the device we are using

//*******  Setup LoRa Test Parameters Here ! ***************

const uint32_t Frequency = 2445000000;                     //frequency of transmissions
const uint32_t Offset = 0;                                 //offset frequency for calibration purposes
const int8_t  TXpower = 3;                                //LoRa transmit power

//*******  Setup LoRa modem parameters here ! ***************
const uint8_t Bandwidth = LORA_BW_1600;                    //LoRa bandwidth
const uint8_t SpreadingFactor = LORA_SF5;                  //LoRa spreading factor
const uint8_t CodeRate = LORA_CR_4_5;                      //LoRa coding rate

//*******  Setup FLRC modem parameters here ! ***************
const uint8_t BandwidthBitRate = FLRC_BR_1_300_BW_1_2;     //FLRC bandwidth and bit rate, 1.3Mbs
//const uint8_t BandwidthBitRate = FLRC_BR_0_260_BW_0_3;   //FLRC 260kbps
const uint8_t CodingRate = FLRC_CR_1_0;                    //FLRC coding rate
const uint8_t BT = RADIO_MOD_SHAPING_BT_1_0;               //FLRC BT
const uint32_t Syncword = 0x01234567;                      //FLRC uses syncword`

Other than the pin definitions, the only other configuration I changed was the TXpower. I changed that from 10 to 3 based on this warning from LilyGo for my dev board:
_"When using the "SX1280 2.4G With PA [H658]", it is recommended that the software control 1280 chip output 2 to 5 dbm, so that the entire module will output 20 dbm of power; When the output of the 1280 chip is higher than 2 dbm, the output power of the entire module is only 20 dbm; When the output of 1280 chip is higher than 5 dbm, the output power of the entire module is only 20 dbm, but if 5 dbm is given to the FEM, it will damage the FEM chip (the maximum allowable input for the FEM is 5 dbm) If the output of the 1280 chip is lower than 2 dbm, the output power of the entire module is within 20 dbm. The output power setting of the SX1280 cannot be greater than 2-5dbm, otherwise, damage to the FEM chip may occur. Set 3dBM to have an output power of 20dmb."_

I am running the program in 'USEFLRC' mode and receiving the below error in the serial monitor:

`
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x42028dc6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core  1 register dump:
PC      : 0x4037c446  PS      : 0x00060036  A0      : 0x4037abc7  A1      : 0x3fce9dd0  
A2      : 0x3fc96134  A3      : 0x3fc96314  A4      : 0x3fc96318  A5      : 0x00000001  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x3fc9613c  A9      : 0x3fce9db0  
A10     : 0x3fcebdd8  A11     : 0x3fcebdd8  A12     : 0x00000014  A13     : 0x00000004  
A14     : 0x3fcebdd0  A15     : 0x80000001  SAR     : 0x0000000e  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x4037c443:0x3fce9dd0 0x4037abc4:0x3fce9e00 0x4037abba:0xa5a5a5a5 |<-CORRUPTED
`

StuartsProjects, 2024-03-03:
> but so far have been unsuccessful in running any of the StuartsProjects/SX12XX-LoRa examples.

Did you start with the simple SX128X examples, '3_LoRa_Transmitter' and '4_LoRa_Receiver' did they work and if not what were the errors ?

TomsGitBits, 2024-03-03:
> > but so far have been unsuccessful in running any of the StuartsProjects/SX12XX-LoRa examples.
> 
> Did you start with the simple SX128X examples, '3_LoRa_Transmitter' and '4_LoRa_Receiver' did they work and if not what were the errors ?

I did start with those programs and have the same issue there. Here's the output of '3_LoRa_Transmitter'

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42024e96
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core  1 register dump:
PC      : 0x4037c446  PS      : 0x00060036  A0      : 0x4037abc7  A1      : 0x3fce9dd0  
A2      : 0x3fc95cec  A3      : 0x3fc95ecc  A4      : 0x3fc95ed0  A5      : 0x00000001  
A6      : 0x00000001  A7      : 0x00000001  A8      : 0x3fc95cf4  A9      : 0x3fce9db0  
A10     : 0x3fcebdd8  A11     : 0x3fcebdd8  A12     : 0x00000014  A13     : 0x00000004  
A14     : 0x3fcebdd0  A15     : 0x80000001  SAR     : 0x0000000e  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x4037c443:0x3fce9dd0 0x4037abc4:0x3fce9e00 0x4037abba:0xa5a5a5a5 |<-CORRUPTED
```

StuartsProjects, 2024-03-04:
Strange that even the startup message "3_LoRa_Transmitter Starting" is not printed. 

With suitable pin number changes the '3_LoRa_Transmitter' sketch works fine on an ESP32S3 Dev Kit board, so there is no issue with the library code. The SX128x code has been un-altered for quite a while now, if the was a problem causing a microcontroller to crash, you would have expected it to have surfaced by now. 

For my ESP32S3 the LoRa device was on the default SPI pins for the dev kit board.

TomsGitBits, 2024-03-04:
Agreed, definitely strange. Though I'd be surprised if the pin definitions were incorrect as I pulled them from a working example as I noted in my initial message. I guess the only 'educated-guess' I made there was that the NSS pin is equivalent to the CS pin. Is that correct?

If you want to sanity check my pin definitions beyond the example provided, here is a link to the [pinout diagrams](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/T3_S3_V1.0.pdf) provided with the board, which I admittedly had a hard time interpreting.

StuartsProjects, 2024-03-04:
Are you sure there are no startup messages printed ?

I would expect on program start to see time, date, version, "3_LoRa_Transmitter Starting" and then "No device responding" if the NSS pin is wrong or the LoRa device is otherwise not found. 

Unfortunatly the library example they have given you, from RadioLib, seems to have info on the the pin numbers for SPI and LoRa internal to that library on a device by device basis. 

There is a pinout diagram and assuming SPI is started with SPI.begin(SCK, MISO, MOSI, NSS); then the SX128X library examples should work. If they dont then its really down to Lilygo to shed some light if there are any config issues with their device.

TomsGitBits, 2024-03-04:
Having read through the program that's what I would've expected as well but I'm not getting anything in the serial monitor other than the error above. I've attached a short video demonstrating this.

https://github.com/StuartsProjects/SX12XX-LoRa/assets/74435159/fa5b7c64-8533-494c-a1ce-9ef955cd42e2

You're correct that the library has those pin definitions listed out by device in the utilities.h file. That is where I originally sourced the pin definitions from and listed in my issue description - under LILYGO_T3_S3_V1_0 ([line 14](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/1c3a46c8f7c9aca856fa0bc071e6192f0461d4cb/examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt/utilities.h#L14C12-L14C29), [line 178](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/1c3a46c8f7c9aca856fa0bc071e6192f0461d4cb/examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt/utilities.h#L178)). Uncommenting line 14 is what enabled that example to work. 

It appears that SPI is started with SPI.begin(SCLK, MISO, MOSI) in the RadioLib example as seen on [line 120 of boards.h](https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/1c3a46c8f7c9aca856fa0bc071e6192f0461d4cb/examples/RadioLibExamples/SX1280/SX128x_Transmit_Interrupt/boards.h#L120). The NSS property doesn't seem to be passed into that function for whatever reason, which is the only difference I can find.

StuartsProjects, 2024-03-05:
I dont think I can assist further, as I have said the library code does work with an ESP32S3 but you appear to be having a configuration issue with that particular LiLygo board which I dont have.

I cannot replicate the pinnout used either, on an ESP32S3 dev kit GPIO36 is used by  the PSRAM.

TomsGitBits, 2024-03-05:
I understand. Thank you for the assistance thus far, I’ll try reaching out to LilyGo to further troubleshoot the configuration issue.

If all else fails, I’d be more than happy to send a board. It could be nice for others interested in this library who don’t have the hardware experience like myself to have a prebuilt option to work with.

TomsGitBits, 2024-03-15:
Good news - I was able to solve the above issue, as well as the issue with the Serial monitor not printing.

On the flip side, I am now running into a new error in running 3_LoRa_Transmitter. I saw a previous issue opened for this error where the individual was using a 5v logic Arduino, but I believe that my LilyGo board has 3.3v logic. Output is as follows:
```
15:07:07 Mar 15 2024
V1.0

3_LoRa_Transmitter Starting
ERROR - Busy Timeout!
ERROR - Busy Timeout!
ERROR - Busy Timeout!
ERROR - Busy Timeout!
ERROR - Busy Timeout!
ERROR - Busy Timeout!
```

StuartsProjects, 2024-03-16:
What was the cause of the Guru Meditation error you reported and what was the solution ?

TomsGitBits, 2024-03-24:
I needed to set **USB CDC On Boot: "Enabled"** in the Arduino IDE Tools menu.

TomsGitBits, 2024-03-25:
I totally understand. It was a fortunate coincidence that LilyGo updated their Readme since I've opened this issue specifically addressing this USB configuration. I hope that my current issue is more library specific now that the IDE is configured properly. 

I have confirmed that the DIO1 pin on my board is set correctly to 9. A couple things I have tried have removed this error but cause other issues.

1. I changed SPI.begin() to SPI.begin(SCLK, MISO, MOSI) in the setup function. I don't quite understand this change as the [SPI docs](https://www.arduino.cc/reference/en/language/functions/communication/spi/begin/) list zero parameters for the SPI.begin() function, however in the RadioLib examples, SPI.begin(SCLK, MISO, MOSI) is used. This change resulted in a **"No device responding"** output.
2. I uncommented SPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0)) on [line 124](https://github.com/StuartsProjects/SX12XX-LoRa/blob/1446034eed59787f5e84b0ed36f4881178dbd462/examples/SX128x_examples/Basics/3_LoRa_Transmitter/3_LoRa_Transmitter.ino#L124). The result of this was that there is no output after **"3_LoRa_Transmitter Starting"** is printed.

StuartsProjects, 2024-03-26:
On an ESP32S3 dev kit board I have, both the 236_SDfile_TransferIRQ and 3_LoRa_Transmitter program work just fine, so safe to assume there is no 'error' in the library. 

However a lot of the Lilygo boards have introduced power management ICs on board which need to be configured to control power to the devices such as LoRa modules etc. This introduces a considerable compatibility problem with a great many libraries. 

So you would need to work out how to configure the power management. I have no plans to introduce it into my LoRa library.

TomsGitBits, 2024-03-26:
I was able to get the programs running successfully. Seemingly, the core change that was needed for the LilyGo board was to change SPI.begin() to SPI.begin(SCLK, MISO, MOSI) in the setup function. 

Thanks for the time.

closed 2024-03-26 by TomsGitBits

StuartsProjects, 2024-03-26:
> Seemingly, the core change that was needed for the LilyGo board was to change SPI.begin() to SPI.begin(SCLK, MISO, MOSI) in the setup function.

But previously you had said;

"I changed SPI.begin() to SPI.begin(SCLK, MISO, MOSI) in the setup function" and it did not work .... This change resulted in a "No device responding" output. ?

TomsGitBits, 2024-03-26:
I think there is something wrong with one of my LilyGo boards. They’re the same model but one of them is giving a "No device responding" output and the other is working as expected whilst running the same program.

AndroidCrypto, 2025-02-09:
> I was able to get the programs running successfully. Seemingly, the core change that was needed for the LilyGo board was to change SPI.begin() to SPI.begin(SCLK, MISO, MOSI) in the setup function.
> 
> Thanks for the time.

@TomGitBits: Thank you for giving this important hint to me as I couldn't find the SX1262 module on my LilyGo T3S3 LoRa board.

Greetings
AndroidCrypto