StuartsProjects/SX12XX-LoRa #81: Compatibility of LoRa Chips: Using SX1262 as Sender and SX1278 as Receiver

khalidalbasheer, 2024-03-23:
I’m working on a project that involves LoRa communication, and I’d like to explore using different LoRa chip models. Specifically, I want to use the Semtech SX1262 as the sender (transmitter) and the SX1278 as the receiver.
My goal is to achieve reliable, long-range communication between these two chips.

Is it feasible to pair the SX1262 and SX1278 in this way?

StuartsProjects, 2024-03-24:
The SX126X and SX127X are compatible from SF7 and above. 

The SX126X normally use a TCXO for frequency control and can be used at lower bandwidths than the SX127X. 

These are limitations of the devices and nothing to do with this library.

khalidalbasheer, 2024-03-24:
So, based on your comment, I can send packets from SX126X to SX127X using your amazing library.
I note that your library settings use SF7.
Is that true?

khalidalbasheer, 2024-03-24:
So, based on your comment, I can send packets from SX126X to SX127X using your amazing library.
I note that your library settings use SF7.
The device I want to use is ESP32 board with LoRa built-in chip produced by Heltec Automation
[image]

StuartsProjects, 2024-03-24:
You adjust the LoRa settings to suit the application.

StuartsProjects, 2024-03-28:
No issues with the library.

closed 2024-03-28 by StuartsProjects