StuartsProjects/SX12XX-LoRa #82: SDfile_Transfer Examples Not Working on ESP32-S3-WROOM-1 And SX1280

TheLittleYellowFiat, 2024-03-24:
Hello, I am working on a project that involves transferring a large array of data from one ESP32 to another using the EBYTE E28-2G4M27SX (based on the SX1280 chip). The data (~4MB) could be held in the ESP's PSRAM but it's currently stored on an SD card for the time being. I wanted to test out the data transfer capabilities with the example sketch but when I tested it out there is no communication between the devices, I just get the "ERROR opening file..." on the transmitter and "SDfile transfer receiver ready" on the receiver side. But I've also tried the 52_FLRC_Transmitter, and 53_FLRC_Receiver examples and they work without an issue on the exact same setup. What am I doing wrong?

Here is the code for transmitter side, I had to change the begin function to `LoRa.begin(NSS, NRESET, RFBUSY, DIO1, RX_EN, TX_EN, LORA_DEVICE)` because the EBYTE module requires the RX_EN and TX_EN Enable pins.

`
#define USELORA                              //enable this define to use LoRa packets
//#define USEFLRC                            //enable this define to use FLRC packets

#include <SPI.h>

#include <SX128XLT.h>
#include <ProgramLT_Definitions.h>
#include "DTSettings.h"                      //LoRa settings etc.

SX128XLT LoRa;                               //create an SX128XLT library instance called LoRa, required by SDtransfer.h

#define ENABLEMONITOR                        //enable monitor prints
#define PRINTSEGMENTNUM                      //enable this define to print segment numbers 
#define ENABLEFILECRC                        //enable this define to uses and show file CRCs
//#define DISABLEPAYLOADCRC                  //enable this define if you want to disable payload CRC checking
//#define DEBUG                              //see additional debug info 

//#define SDLIB                              //define SDLIB for SD.h or SDFATLIB for SDfat.h
#define SDFATLIB

#include <DTSDlibrary.h>                     //library of SD functions
#include <SDtransfer.h>                      //library of data transfer functions

//choice of files to send
//char FileName[] = "/$50SATL.JPG";          //file length 63091 bytes, file CRC 0x59CE
char FileName[] = "/$50SATS.JPG";        //file length 6880 bytes, file CRC 0x0281
//char FileName[] = "/$50SATT.JPG";        //file length 1068 bytes, file CRC 0x6A02

void loop()
{
  uint32_t filelength;

#ifdef ENABLEMONITOR
  Monitorport.println(F("Transfer started"));
#endif

  filelength = SDsendFile(FileName, sizeof(FileName));

  if (filelength)
  {
#ifdef ENABLEMONITOR
    Monitorport.println(F("Transfer finished"));
#endif
  }
  else
  {
#ifdef ENABLEMONITOR
    Monitorport.println(F("Transfer failed"));
    Monitorport.println();
#endif
  }

  delay(15000);

}

void led_Flash(uint16_t flashes, uint16_t delaymS)
{
  uint16_t index;
  for (index = 1; index <= flashes; index++)
  {
    digitalWrite(LED1, HIGH);
    delay(delaymS);
    digitalWrite(LED1, LOW);
    delay(delaymS);
[... 358 lines cut ...]
const uint32_t FunctionDelaymS = 0;             //delay between functions such as open file, send segments etc
const uint32_t PacketDelaymS = 1000;            //mS delay between transmitted packets such as DTInfo etc

const uint8_t StartAttempts = 2;                //number of attempts to start transfer before a fail
const uint8_t SendAttempts = 5;                 //number of attempts carrying out a process before a restart
const uint32_t NoAckCountLimit = 250;           //if no NoAckCount exceeds this value - restart transfer

const uint8_t HeaderSizeMax = 12;               //max size of header in bytes, minimum size is 7 bytes
const uint8_t DataSizeMax = 245;                //max size of data array in bytes
const uint8_t Maxfilenamesize = 32;             //size of DTfilename buffer

const uint16_t NetworkID = 0x3210;              //a unique identifier to go out with packet

#ifdef USELORA
const uint8_t SegmentSize = 245;                //number of bytes in each segment, 245 is maximum value for LoRa
#endif

#ifdef USEFLRC
const uint8_t SegmentSize = 117;                //number of bytes in each segment, 117 is maximum value for FLRC
#endif
`

I also tried using FLRC and nothing changes. I am out of ideas so I would greatly appreciate an idea on what to try next.

P.S. Sorry for the bad formatting I had a bit of an issue with markdown.

StuartsProjects, 2024-03-25:
Do the 3_LoRa_Transmitter and 4_LoRa_Receiver examples transmit and receive ?

And what SD card holder are you using ?

TheLittleYellowFiat, 2024-03-25:
Yes, those examples also work. I am using an ordinary micro SD to SD card adapter with wires soldered to it as the SD card holder, but I do not think that's the issue as it initializes the card just fine, and another example sketch lists the complete directory of the SD card. If you have an example that stores the data to be sent in PSRAM I would gladly give it a try, infact that would be closer to my end goal, as I would like to use the ESP's builtin PSRAM instead of the SD card to store the data.

StuartsProjects, 2024-03-25:
> Yes, those examples also work. I am using an ordinary micro SD to SD card adapter with wires soldered to it as the SD card holder, but I do not think that's the issue as it initializes the card just fine, and another example sketch lists the complete directory of the SD card.

Yes, but I asked for a reason, the most commonly used SD card adapter used by Arduionoists, has a major design flaw, it causes problems when used on the same SPI bus as another SPI device. 

When I use an SD card on an ESP32, all varients, I normally put it on a seperate SPI bus, more reliable that way. This seems to be a generic 'issue' with the Expressif ESP32 core, nothing I can do about it. The SD card seems to be more relaible in MMC mode too. 

Whilst the 241_StuartCAM_ESP32CAM_LoRa_Receiver should receive ang check a file into PSRAM without an SD card being active, the transmitter still needs to get the file\image from somewhere, not everyone is using ESP32CAMS, . There is a specific reason for using SD cards, the programs become generic, and can be used on most all Arduinos. Relying on RAM buffers would require a host of different examples and testing.

TheLittleYellowFiat, 2024-03-25:
Understandable. I tried the 241_StuartCAM_ESP32CAM_LoRa_Receiver example with the 233_SDfile_Transfer_Transmitter, and there is communication which is a huge step forward. On the reciever I get `/$50SATS.JPG Array start write request` followed by `/$50SATS.JPG Source array length is 13 bytes` `Source array CRC is 0xAA7E` and unfortunately after that it resets with this message: `rst:0x8 (TG1WDT_SYS_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)`

On the transmitter side I get this:
`Using LoRa packets

Initializing SD card...SD Card initialized.

Payload CRC enabled
SDfile transfer ready

Transfer started
Send file attempt 1
Start file transfer for /$50SATS.JPG
Send open remote file request
/$50SATS.JPG opened OK on remote
Source file length 13 bytes
Source file CRC is 0xAA7E
Segment Size 245
Number segments 1
Last segment size 13
Last segment
0
NoACK
NoACK
NoACK
NoACK
NoACK

************************
/$50SATS.JPG
ERROR in SDsendSegments()
Restarting transfer
***********************

Send file attempt 2
Start file transfer for /$50SATS.JPG
Send open remote file request
NoACK

Send open remote file request
NoACK

Send open remote file request
NoACK

Send open remote file request
NoACK

Send open remote file request
NoACK

********************
/$50SATS.JPG
ERROR opening file
Restarting transfer
******************** `

StuartsProjects, 2024-03-26:
I just ran a test on an ESP32S3 Dev Kit board I have, with the ESP32S3 first running 233_SDfile_Transfer_Transmitter and then 234_SDfile_Transfer_Receiver examples.

No problems when the pins were changed in the examples as required.

With a 65536 bye long file transfer time was 5.04 seconds in LoRa mode and 2.85 seconds in FLRC mode.

TheLittleYellowFiat, 2024-03-26:
Hmmm, well that's interesting. Can you share what pins you used? Also, do you use Ebyte's LoRa modules or some other modules?

StuartsProjects, 2024-03-27:
There was a large amount of testing of those routines done over several months. 

I was using eByte E282G4M20S modules.  

#define NSS 10
#define RFBUSY 21
#define NRESET 1
#define DIO1 2
#define RX_EN 42
#define TX_EN 5

#define SCK 12
#define MISO 13
#define MOSI 11

#define SDCS 4

With ESP32 modules there are now a great many varients, with no overall standard of which pins are the 'default' SPI ones. Its just not practical to cover and keep up to date with all these varients. Whilst on the ESP32 you can re-assign the SCK, MOSI and MISO, there are a great many caveats as to which pins you cannot use. Providing detail on these 'not to use pins' is out of scope for this library.

TheLittleYellowFiat, 2024-03-27:
Well, it's working now. I connected everything to the pins you specified and it transferred a file without any problems. Looks like I used some pins which I shouldn't have, I checked with this [http://wiki.fluidnc.com/en/hardware/ESP32-S3_Pin_Reference](url) ESP32-S3 pin reference before wiring it the first time (I'm leaving the link here if someone happens to have the same issue they might find it helpful), but I've somehow messed it up anyway. I also tried the 241_StuartCAM_ESP32CAM_LoRa_Receiver example again, but it still crashes when it tries to receive. Now I'm going to try to figure out how to send a large array from one ESP's PSRAM to the other's PSRAM.

Thank you for the very quick support and all the very helpful information.