StuartsProjects/SX12XX-LoRa #84: TXpower - Maximum power SX128

Nicolesilvaa, 2024-04-05:
The G-niceRF has a LoRa radio with a power of 26.3 dBm. Can I use this module in this library? Or is the maximum power 10 dbm?

StuartsProjects, 2024-04-05:
The SX128x device can only be set for a power of -18dBm to +12dBm. 

NiceRF will have added a power amplifier to the SX128X, so there may be no external control of how much power the module is putting out. 

You would need to check the devices datasheet for the actual power levels the module produces, when for example the SX128x is set to 10dBm.

StuartsProjects, 2024-04-05:
Well presumably NiceRF intend you to drive the SX128X set to its max power level or 13 to get the 26.4dBm output. 

Whether you get that level of output, I cannot be sure, I have not tested that module before.

Nicolesilvaa, 2024-04-05:
Okay. Thanks 😊

StuartsProjects, 2024-04-20:
No issues with the library, the module manufacturer would need to advice on the power settings for this module.

closed 2024-04-20 by StuartsProjects