StuartsProjects/SX12XX-LoRa #85: Frequency Incorrect 915000000hz

AdamHawkTrack, 2024-04-07:
Hi Stuart, I am trying to use your example code 23 tracker and 24 receiver for SX126X. The transmitter seems to be working fine but not the receiver. Despite the settings.h frequency being set to 434000000, the serial monitor shows 915000000hz. I'm not sure when it's getting that from. Any help would be appreciated. Kind regards, Adam

StuartsProjects, 2024-04-08:
The frequency setting code is the same for transmitter and receiver. Example 24 receiver works OK here, correctly reporting being set to 434000000. 

915000000hz is the frequenct displayed from reset, the default. 

Which SX126X module are you using and which Arduino ?

Try examples 103 and 104, do they work together ?

AdamHawkTrack, 2024-04-08:
I'm using the ebyte e220-400mm22s which has the LLCC68 chip. It's connected to a Pro Mini.

I tried the 103 and 104 examples and I'm getting exactly the same result with regards to the frequency.

I can get the device's to communicate on frequency 434 using the lora-rf code, but I'm struggling to send the GPS coordinates, so was hoping your code might work for me.

StuartsProjects, 2024-04-08:
I have tried the Library against a NiceRF LLCC68, and it worked, so there is no inherrant issue with the library. 

However, there can be differances in the setup required, in particular related to the TCXO and regulator setups, the library has no way of determining how a particular manufactures modules have been designed. See example 103 for the breakdown of the device settings, in particular;

setRegulatorMode(USE_DCDC);
setDIO3AsTCXOCtrl(TCXO_CTRL_3_3V);      
setDIO2AsRfSwitchCtrl();

Presumably eByte have provided the setup details for thier modules ?

>  I can get the device's to communicate on frequency 434 using the lora-rf code, but I'm struggling to send the GPS coordinates

I dont uderstand, what is the 'lora-rf' code ?

Can you clarify which of the library examples work and which do not ?

AdamHawkTrack, 2024-04-08:
[E220-400MM22S_UserManual_EN_v1.3.pdf](https://github.com/StuartsProjects/SX12XX-LoRa/files/14907921/E220-400MM22S_UserManual_EN_v1.3.pdf)

Sorry I wasn't clear. I tried the LoRa-RF Arduino Library by chandrawi. The examples work for sending messages but I'm struggling to send the GPS lat long. On that library it works when I commented the TXCO clock reference and used XTAL.

Maybe you library isn't compatible. It just seems odd where it's getting the 915000000hz from.

I am a beginner to code and most of what I've achieved to date is using libraries and parts of code that others have shared, so please bare this in mind.

Thanks

StuartsProjects, 2024-04-08:
The 24 and 25 examples you tried use SF12 and as I recall the LLCC68 only supports up to SF11, there is a note in the documentation about these potential issues;

_"The newer LLCC68 LoRa modules should work when programmed as SX126X devices although check the data sheets for the range of LoRa modem parameters supported"_

Its just not practical for the library to automatically identify all the incompatibilities with how the LoRa modules are configured.

The reporting of a frequency of 915000000hz might be an incompatibilty between SX126X and LLCC68. The SX126X does not have a documented way of reading back the frequency its been set to, I implemented a workaround and its possible it does not work on the LLCC68.

However I still dont know if the 103 and 104 examples actually worked as transmitters and receivers .......................

StuartsProjects, 2024-04-08:
The NiceRF LLCC68 I have tested does display the frequency correctly.

But as expected it does not work at SF12, so the LoRa settings in examples 24 and 25 would need to be changed.

AdamHawkTrack, 2024-04-08:
Yes, you're correct I'd overlooked the LLCC68 spread factor. I have changed this and the bandwidth, but still no success. The 103 example looks like it's sent the data successfully, but the receiver just says it's ready. 

I noticed an odd thing. When the receiver shows 915000000hz in the serial monitor and I press the reset button on the pro mini it will change to 434000000hz. Then I noticed it does the same if I close and reopen the serial monitor, but doesn't receive anything.

StuartsProjects, 2024-04-09:
There is nothing that suggests there is any inherrent problem with the library and LLCC68 devices. I can use a NiceRF LLCC68 just fine with examples 103 and 104 when the TCXO setting option is removed, Some LLCC68s dont have TCXOs to reduce costs. 

That your haveing problems suggests some form of wiring problem, but I have not seen a full list of the connections you are using. 

Is it a 5V\16Mhz Arduino Pro Mini you have ?

In general the LLC68 without an TCXO is not an optimum choice for a tracker, as you cannot use SF12 for long range or drop below bandwidths of 125khz on 868Mhz or 62.5khz on 434Mhz.

AdamHawkTrack, 2024-04-09:
I'm getting a little exhausted trying to work with this module. I chose to work with this because of the small size and distance the the module claims to have. Perhaps due to its small size it's absent of certain hardware.

I'm using the 3.3v 8mhz pro mini as both the lora module and gps module work at that voltage.

I think I'm going to order some NiceRF modules and see how they work me 

I'm not sure how much bandwidth affects the range, but I thought the lower frequency was better for conditions when it's not a clear line of sight.

StuartsProjects, 2024-04-20:
No issues were found with the library.

closed 2024-04-20 by StuartsProjects