StuartsProjects/SX12XX-LoRa #87: Example of lora sx1278 module

Pipoteex, 2024-04-20:
Hi StuartsProjects! I hope you are fine!

I was reading some examples of applications you made, they are really great!

I would like to know if you could help me with a code for lora.
I have 3 SX1278 RA-02 433MHZ modules and I need to be able to communicate those three modules between them.

How could I make sure that packages are sent between the two?

I need each lora to be able to receive packets from different loras, and through a code detect where it comes from and thus forward that information to another lora module.

I thank you in advance!

StuartsProjects, 2024-04-20:
I cannot supply support for individual user applications, I only provide support for potential issues\bugs with the library code and examples provided. 

Review the examples, there are ones that deal with addresssing packets to particular nodes and requireing acknowledgements from the nodes for transmitted packets.

StuartsProjects, 2024-04-23:
There are examples of using addressing of packets, such as those in the Relaible folder. 217_Reliable_Transmitter_Data_Requestor for isntance shows how to send a request to a particular node and request a response. See the build of the packet from line 107 onwards. 

Addressing packets is not a LoRa issue as such, it just depends on what bytes you add to a packet and what the receiver does when it receives the packet.

closed 2024-04-23 by StuartsProjects

Pipoteex, 2024-04-23:
Great, I was looking at the examples and the truth is that they are very good!

Thank you so much!

Pipoteex, 2024-05-30:
Hello! How are you? I wanted to ask you, if the function: LT.transmitReliable(RXBUFFER, RXBUFFER_SIZE, NetworkID2, RXtimeout, TXpower, WAIT_TX)

can send a string, it's just that it's difficult for me to work with uint8_t RXBUFFER[251]

because what I need is to be able to concatenate data to a package.

For example: Lora 1 sends packet: R1-A1 and that packet is received by Lora 2 and concatenates to that same packet another message such as R0-A0 and sends to Lora 3. Lora 3 should receive R1-A1R0-A0, but Due to the type of data, I don't know how to do the concatenation correctly.

I hope you can give me a hand.

StuartsProjects, 2024-05-31:
The library does not support direct string handling. 

In general the use of strings on Arduinos is not recommended, it can cause memory fragmentation problems. 

You would need to copy the string to a character array and send that.

StuartsProjects, 2024-06-12:
> I want to ask you what the RXTIMEOUT and WAIT_RX variables are used for. 

There is an explanation in the 'Receiving Packets' section of the document you see on the main page of the GITHUB repository