StuartsProjects/SX12XX-LoRa #89: Really bad range SX1280

Vincent-NB, 2024-05-06:
I'm attempting to use this library with some Lilygo T3-S3 boards but I'm only getting a couple meters of range, even using spreading factor 11 at 203 kHz.

I've tried the example code provided by Lilygo with the same settings and I get over 200m in an urban environment with the same boards. 
I'm using the boards with built in power amplifier so I've set the TXpower to 3 dBm per Lilygo's instructions. 

I'm using a modified version of the 104_LoRa_Receaver_Detailed_Setup_ESP32 for testing with the only modification being that it also shows the serial monitor output on the oled display.

Regardless of the distance I can't get any RSSI  value over -80 whereas with the Lilygo code it starts around -40
Using a different ESP32 dev-board with a LAMBDA80C-24D lora module I get an RSSI of -37 at the same distance (with a slightly different antenna.)

With all of these I'm using the same Lilygo board as the transmitter so I think it has to be something on the receiver side.

StuartsProjects, 2024-05-06:
The library has not been specifically tested against the 'Lilygo T3-S3' boards and there do seem to be a few variations. 

Unfortunatly the manufacturers of boards do not donate them to be tested against the library. 

You said the boards have 'built in power amplifier' which suggests that need specific, as in not SX1280 module standard, switching between TX and RX modes. The library does support a standard form of this switching but you need to know the GPIO pins involved. 

Maybe ask Lilygo as to what GPIO pins need to be set for RX mode and which for TX mode ?

Vincent-NB, 2024-05-06:
Is that what the TX_EN and RX_EN macro's do?
They are in the 103_LoRa_Transmitter_Detailed_Setup_ESP32 settings file but not in the 104_LoRa_Receaver_Detailed_Setup_ESP32. Would it be as simple as putting the RX_EN pin high during setup on the receiver.

Vincent-NB, 2024-05-06:
I've also been unsuccessful at trying any other example with this board. All except the ESP32 examples result in a "ERROR - Busy Timeout!"

StuartsProjects, 2024-05-07:
"ERROR - Busy Timeout!" means the RFBUSY pin cannot be read correctly. This occurs if the pin given for RFBUSY is not correct.

closed 2024-05-09 by Vincent-NB