StuartsProjects/SX12XX-LoRa #90: Continuous Lora modulated signal at 500Khz bandwidth?

chrisdimoff, 2024-05-12:
I am getting ready to bring an electronic device using SX1276 Lora radio to market. The FCC testing lab has requested that I be able to put the radio into the following configuration:

100% Duty Cycle modulated signal with 500 kHz BW
I cannot figure out how to do a Lora modulated signal (with 500 khz BW) at a 100% duty cycle.

The best I can do so far is:

a 100% duty cycle/continuous signal with FSK modulation (bandwidth is not 500khz, no lora settings, etc). I can do this by using this code: https://github.com/StuartsProjects/SX12XX-LoRa/blob/master/examples/SX127x_examples/Basics/80_Direct_DIO2_Tone/80_Direct_DIO2_Tone.ino

a lora modulated signal with correct parameters that is NOT 100% duty cycle. I have to send packets in a loop and the signal/wave is "bursty" because of this. I can do this using this code with zero delay between loop iterations: https://github.com/StuartsProjects/SX12XX-LoRa/blob/master/examples/SX127x_examples/Basics/103_Lora_Transmitter_Detailed_Setup/103_Lora_Transmitter_Detailed_Setup.ino

How can I get a continuous, Lora modulated signal with lora settings (bandwidth, sf, etc) that has a 100% duty cycle?

Thank you

StuartsProjects, 2024-05-13:
There is a register setting, listed in the datasheet, that forces LoRa transmit to go into a continuous mode. However, there are no examples provided that support this feature. 

You need to approach the manufacturer of the LoRa boards you are using for assistance, for obvious reasons I cannot make claims that my free library is suitable for FCC compliance testing. 

I do not do paid consultancy.

chrisdimoff, 2024-05-13:
@StuartsProjects thank you for the response! 

I was able to find the register that puts it into continuous mode, but it seems that it only transmits in continuous mode for the  FSK modem and not the LORA modem? I cannot get a lora modulated continuous mode... only FSK. 

Semtech has been less than helpful for me here - i have reached out a few times to them. 

I 100% understand that your library makes no claims for FCC testing - no worries there. 

I am reaching out because I am running out of options and you and this project seem to have a very good understanding of the sx1276 and I thought someone here could help me. 

I am looking for someone to help me identify the specific registers, and in what order, to get a lora modulated 500khz continuous wave. I can post code snippets I have tried here if that is helpful. 

Thanks

StuartsProjects, 2024-05-14:
> I am looking for someone to help me identify the specific registers, and in what order, to get a lora modulated 500khz continuous wave.

If you are developing a commercial product and you dont understand the information in the device datasheet, or how to implement the changes you want, then clearly you either need to employ an approprate designer or ask whoever you are paying for the modules for support.

netweaver1970, 2024-05-14:
> @StuartsProjects thank you for the response!
>
> I was able to find the register that puts it into continuous mode, but it seems that it only transmits in continuous mode for the FSK modem and not the [...]

It IS possible to put the SX1276 in Lora continuous transmit mode (straight search from the Datasheet, even with reference to the spectral/EMC testing): 
[image]

But I can't help with code for this. Anyway, look for the "TxContinuousMode" register. Maybe the (incomplete) MicroPython code at this github can point you in a direction, as it clearly mentions this register, both possible values + register setting sequence. https://github.com/xg590/SX1276/blob/main/lora.py

[image]

There's not exactly a lot of immediately usable info on the internet on this, for you to just apply. But you should be able to figure it out, combining all bits of info, as there are bits and pieces around on this, always around EMC testing.

StuartsProjects, 2024-05-14:
I have no plans or intention to introduce changes or code to the library that might be required for regulation compliance testing.

closed 2024-05-14 by StuartsProjects