StuartsProjects/SX12XX-LoRa #92: Support for MSPM0 Microcontroller

rajeshchiplogictech, 2024-05-29:
I would like to implement this library on the MSPM0 microcontroller using CCStudio. Could you please provide guidance or support on how to adapt this library for use with the MSPM0 board? Specifically, I need help with the following:

Platform Definitions: How can I define the MSPM0 platform correctly in the library?
Integration Steps: Are there any specific steps or modifications needed to integrate the library with CCStudio and the MSPM0 microcontroller?
Example Code: If possible, could you provide an example code snippet demonstrating how to initialize and use the library with the MSPM0 microcontroller?

StuartsProjects, 2024-05-29:
I have no experience of using the MSPM0 microcontrollers or CCStudio, so I cannot help you.

StuartsProjects, 2024-05-30:
No support for MSPM0 or CCStudio.

closed 2024-05-30 by StuartsProjects