StuartsProjects/SX12XX-LoRa #95: SX1278 esp32Cam getting the transmitters battery voltage

perrer12, 2024-07-31:
Hello,

i use the ESP32Cam as Transmitter an an ESP32 Wrover as Receiver to transmit a picture from the examples, which works fine.

I read the Battery Voltage on Pin 14 of the ESP32cam and tried to transmit this after the picture was succesfully sent.

For this, i used the simple LoRa.transmit and LoRa.receive from the basic examples, but after receiving the picture, i get a timeout und no data was received.

Whats the best way to send the battery voltage to the receiver after receiving the picture ?

Thanks in advance

Perrer

perrer12, 2024-07-31:
I also use LittleFS for transmitter and receiver, so i got Pin 14 for analog read

StuartsProjects, 2024-08-01:
LoRa.transmit() and LoRa.receive() functions use DIO0 to detect TX and RX complete. 

But the ESP32CAM does not use DIO0, which is why all the transfer code uses the IRQ versions of transmit and receive. 

See the examples 3_LoRa_TransmitterIRQ and 4_LoRa_ReceiverIRQ

perrer12, 2024-08-01:
With using IRQ, it works now.

Thank you very much !

closed 2024-08-01 by perrer12