StuartsProjects/SX12XX-LoRa #96: setDIO3AsTCXOCtrl set as 3.3V in the library

floatAsNeeded, 2024-09-03:
I noticed the command setDIO3AsTCXOCtrl in the setup LoRa always sets the TCXO to 3.3V, which in the NiceRF the TXCO voltage is usually 2.8V and in the Dorji is around 1.8V. Maybe better to define the voltage of TCXO separately in the examples, depending on the module used.

Any thoughts?

StuartsProjects, 2024-09-04:
You can change the TCXO voltage, see the details in the '103_LoRa_Transmitter_Detailed_Setup' example. 

Its not been an issue (needing to set a different TCXO Voltage) until recently when I have noticed that some SX126x modules are now being produced with no TCXO at all. So I have been looking at a simple way of turning off the setDIO3asTCXOCtrl config. 

In addition you cannot just use the setDIO3AsTCXOCtrl as you need to do the calibration afterwards too, as shown in the '103_LoRa_Transmitter_Detailed_Setup' example. 

Current thinking is to add a variant of the setDIO3AsTCXOCtrl function that allows the user to enable calibration, then it would be up to the user to reference what TCXO voltage a particular module needs.

floatAsNeeded, 2024-09-04:
Oh okay, instead of using the setupLoRa function, you can just call all the required functions and adjust the TCXO voltage as well. Thanks for pointing out that example.

I agree with the variant proposed and maybe directly in the setupLoRa function or maybe another detailed setupLora function. It could be added another parameter for the TCXO voltage in particular or something to disable it. 

For example, to define a disable value for the TCXO voltage in the h file

#define TCXO_CTRL_DISABLE 0xFF

and then the setupLoRa (or another function) can also have the parameter for the TCXO and with an if statement to disable 

void SX126XLT::setupLoRa(uint32_t frequency, int32_t offset, uint8_t modParam1, uint8_t modParam2, uint8_t modParam3, uint8_t modParam4, uint8_t tcxoVoltage)

if (tcxoVoltage != TCXO_CTRL_DISABLE) {
    setDIO3AsTCXOCtrl(tcxoVoltage);   
    calibrateDevice(ALLDevices);         
  }