StuartsProjects/SX12XX-LoRa #98: Readbuffer on ISR

simon88, 2024-10-21:
Hi 
I'm trying to create ping pong non blocking mode but I have an issue when I'm trying to call `LT.readBuffer`

Here is my ISR function 
```
void ISR1() {
  DEBUG_PRINTLN("DIO1 is HIGH!!");
  uint16_t IRQStatus;

  IRQStatus = LT.readIrqStatus();

  if (IRQStatus & IRQ_TX_DONE) {

    LT.receiveSXBuffer(0, 0, NO_WAIT);  // according to IRQ register TX is done so we can put the module into receiving mode
    DEBUG_PRINTLN("Module in receiving mode");
  }
  if (IRQStatus & IRQ_RX_DONE) {

    uint16_t Packetcode;

    LT.startReadSXBuffer(0);
    Packetcode = LT.readUint16();  //here I have receive data in Packetcode how to print data ??
    LT.readBuffer(recv_buffer); // my board stuck here
    LT.endReadSXBuffer();
  }
}

```
My isr function works well but I'm stuck to read and print data

StuartsProjects, 2024-10-21:
Thats a lot of code for an ISR, you normally keep them very short, just set a flag etc and no printing etc.  

LT.readBuffer(recv_buffer) is assuming its reading a character buffer so keeps reading till it finds a null.

simon88, 2024-10-22:
thanks for you comment @StuartsProjects . I've updated the code like that but I still have the issue
````
uint8_t receivebuffer[16];
bool recv = false;

void ISR1() {
  uint16_t IRQStatus;
  IRQStatus = LT.readIrqStatus();

  if (IRQStatus & IRQ_TX_DONE) {
    LT.receiveSXBuffer(0, 0, NO_WAIT);
  }
  if (IRQStatus & IRQ_RX_DONE) {
    recv = true;
  }
}
````

and in my loop

````
void loop(){
  if(recv){
    LT.startReadSXBuffer(0);
    LT.readBuffer(receivebuffer);  //stuck here 
    LT.endReadSXBuffer();
    recv = false;
  }
}
````

I don't understand why if I call `readBuffer` my code stuck, because if I cal for example `readUint16` I don't have issue. 

For information in my tansmitter I send like this:
````
uint8_t hello[] = "HELLO";
LT.transmit(hello, sizeof(hello), 0, TXpower, WAIT_TX)
````

StuartsProjects, 2024-10-22:
I cannot comment on whether the code you have written will work inside an ISR, I would normally just set a flag in the ISR and process the packet in loop() as normal. 

I suggest you write the program without the ISR to check your code. There is an example of  using LT.readBuffer() which does work.

StuartsProjects, 2024-11-09:
Issue assumed closed.

closed 2024-11-09 by StuartsProjects