StuartsProjects/SX12XX-LoRa #99: Images Missing From SX12XX-LoRa/examples/SX128x_examples/Ranging

ElectronicallyE, 2024-10-22:
Hi Stuart,

Looks like the [linked](https://github.com/StuartsProjects/SX12XX-LoRa/tree/master)/[examples](https://github.com/StuartsProjects/SX12XX-LoRa/tree/master/examples)/[SX128x_examples](https://github.com/StuartsProjects/SX12XX-LoRa/tree/master/examples/SX128x_examples)
/Ranging images on this page are missing (404) error. Might want to fix when you have the opportunity?

StuartsProjects, 2024-10-22:
The images were in the GITHUB, but it appears an update changed the address of the link. Its been corrected. 

Thanks.

closed 2024-10-22 by StuartsProjects

ElectronicallyE, 2024-10-23:
Hi Stuart,

Thanks for rectifying.

Looks like this image is still giving a 404 error:
[examples/SX128x_examples/Ranging/Pictures/SX128XLT_Ranging_100m.jpg](https://github.com/StuartsProjects/SX12XX-LoRa/blob/master/examples/SX128x_examples/Ranging/Pictures/SX128XLT_Ranging_100m.jpg)