StuartsProjects/SX12XX-LoRa #100: Set SetRxDutyCycle mode possible?

jrhbcn, 2024-11-09:
Hi,

First of all, thank you the library! I am trying to create a program very similar to the wake_on_rx. However, to save power, I am trying to set up the SetRxDutyCycle so the radio sleeps for some time without losing packages (if the timing is set correctly). I don't seem to find any function to set this. 

It is not implemented? Is there a way to set it and keep using the library?  My guess is that I should probably modify the function `RXPacketL = LT.receive(RXBUFFER, RXBUFFER_SIZE, 0, NO_WAIT); `but maybe I can set something up before and there is no need to modify it... :?

StuartsProjects, 2024-11-09:
There are no library functions that implement the SetRXDutyCycle mode. 

I have never tried to implement it, so cannot really assist you. 

It looks like a library modification is needed, since currently the receive library functions use the setRX() function and I assume that would need modifying.

StuartsProjects, 2025-02-24:
Not supported by library

closed 2025-02-24 by StuartsProjects