StuartsProjects/SX12XX-LoRa #102: No response from DIO1

GrimmsterZed, 2025-01-19:
Hello!

First I want to say, thank you for writing such a thorough LoRa library. It is easy to follow along with the Datasheet.

Unfortunately I seem to be in the same place as [Issue 16](https://github.com/StuartsProjects/SX12XX-LoRa/issues/16), where I am never getting a high voltage out of DIO1 using the ESP32-S2. I am using the Basic Transmitter Detailed Setup program, and with some wolf fencing I figured out it's waiting for DIO1 to rise to finish transmission. With a logic analyzer I can confirm that the pin is never rising.

My current setup is the [ESP32-S2-DevkitC-1](https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s2/esp32-s2-devkitc-1/user_guide.html) and the WIO SX1262 from [this kit](https://www.seeedstudio.com/Wio-SX1262-with-XIAO-ESP32S3-p-5982.html), using breakout headers instead of the connector to the provided XIAO ESP32.

I was wondering if since that issue you have had discovered anything about using the ESP32-S2 with this chip, or have obtained any other insight on my issue. I can provide further details and provide logic analyzer outputs if needed

Thank you!

StuartsProjects, 2025-01-19:
Issue 16 was related to the modules being used, not the ESP32S2, ni issues with the library were found. I don't have the ESP32S2 module you describe, but have used the library extensively with ESP32 and ESP32S3, no issues found. 

Take a look at program 103_LoRa_Transmitter_Detailed_Setup, which shows the full detailed setup for the SX1262. 

Whilst most SX1262s will work with the setup below for the TCXO crystal oscillator;

LT.setDIO3AsTCXOCtrl(TCXO_CTRL_3_3V);

Some SX1262 modules may need that config step removed, or the voltage changed. If the setting is wrong then the modules can hang at transmit or receive. You need to ask the manufacturer of the device what settings are needed for their device.

StuartsProjects, 2025-01-19:
The possible voltage settings for the TCXO are in the libraries SX126XLT_Definitions.h file;

TCXO_CTRL_1_6V
TCXO_CTRL_1_7V
TCXO_CTRL_1_8V
TCXO_CTRL_2_2V
TCXO_CTRL_2_4V
TCXO_CTRL_2_7V
TCXO_CTRL_3_0V
TCXO_CTRL_3_3V

GrimmsterZed, 2025-01-20:
Hello!

Thank you for the thorough response!

I gave altering the TCXO voltage a try which unfortunately didn't work. However, I did scrape up a pair of RA-01SH modules which ended up working for me without alteration.

Seeed's documentation of their module is quite crappy. They don't provide a proper schematic and don't specify a TCXO voltage to use. I've submitted a ticket to get more information from Seeed themselves about startup configuration.

Thanks again for your help!

closed 2025-01-20 by GrimmsterZed