StuartsProjects/SX12XX-LoRa #103: Probably interoperability issues between SX1262 and SX1276 when using Reliable sketches

AndroidCrypto, 2025-01-19:
Dear Stuart,

First of all, let me say "thank you" for your LoRa library that allows me to use the (nearly) same sketch for different hardware configurations.

**I'm facing the problem that a device with a SX1262 module on receiver side does not recognize/act on a transmit from a SX1276 module on sender side in Reliable mode.**

This is the hardware I'm using for my tests:

a) Heltec Wi-Fi LoRa 32 V2.1 board (that is an ESP32, SX1276 LoRa module and an OLED display SSD1306)
b) Heltec Wi-Fi LoRa 32 V3.1 board (that is an ESP32-S3, SX1262 LoRa module and an OLED display SSD1306)

I'm using a fixed frequency, Spreading Factor and Coding Rate in the 868 MHz band (Europe).

I started my tests, as recommended by you, with the '2_Register_Test.ino' examples with success. Then I run the 
'17_Sensor_Transmitter.ino' and '18_Sensor_Receiver.ino' pair with success as well. I changed the roles of Heltec V2.1 
and Heltec V3.1 boards, and I was being able to transfer sensor data between both devices.

The problem raised when I tried to run the **'Reliable' sketch pair** '215_Reliable_Transmitter_ACK_withData.ino' and 
'216_Reliable_Receiver_ACK_withData.ino'. When the Heltec V 2.1 board (SX1276) acts as a sender (transmitter) the 
Heltec V 3.1 board (SX1262) **does not recognize any message**. After changing the roles (Heltec V 3.1 is the sender, Heltec V 2.1 is the receiver) all tests worked as expected.

During my tests, the Heltec V 3.1 board sometimes received the packet and responded correctly with an ACK. The next transmission didn't work as be before. So my suggestion is that any (internal) register, flag or IRQ setting is most of the time in a wrong state, but due to an earlier transmission it is set in a way the sketch does recognize the incoming packet.

I do not add any sketches because it is a lot of code, but of course I would provide them if they are helpful for you. 
For any communication outside of GitHub please use my regular email address androidcrypto@gmx.de.

Thanks in advance for your help, as actually I'm writing an ESP32 based tutorial about LoRa and I would like to recommend your library.

Thanks in advance, kind regards,
Michael

AndroidCrypto, 2025-01-20:
Stuart,

In issue #102 ("No response from DIO1") I read about "hanging" transmissions or receipts and I tried to use this additional setting after the regular "if (LT.begin(NSS, NRESET, RFBUSY, DIO1, LORA_DEVICE)) " call:

**LT.setDIO3AsTCXOCtrl(TCXO_CTRL_1_8V);**

A short test gave positive results, however, can you recommend using this setting ?

Michael

StuartsProjects, 2025-01-20:
I can find no interoperability issues, that was with sketches 215 and 216 on an ESP32S3\SX1276 and  Heltec Wi-Fi LoRa 32 V3.1.

Are you using those sketches **exactly as published** with no changes\edits at all ?

AndroidCrypto, 2025-01-21:
Stuart, after some extended tests, I found that the Heltec Wi-Fi LoRa 32 V3.1 boards are a bit different from the implementation in your published examples. Why the issue showed up only when using the "Reliable" receiver sketch is unknown to me, but I will publish my workaround as a help for others.

I found an [post](http://community.heltec.cn/t/wifi-lora-32-v3-1-tcxo-circuit/16906) in the Heltec community forum:

```
Q: ... SX1262 chip has a dedicated regulator for TCXO that is used to power an actual TCXO on the PCB. This regulator is programmatically adjustable, what voltage should be set to power TCXO correctly?

A:  I have found the value of TCXO voltage in Meshtastic firmware, it is 1.8V.

```

As my code should be used for different boards, I'm using compiler directives for the dedicated board (here for the Heltec V3 one in setup()), just place the LT.setDIO3AsTCXOCtrl call after the 'begin' sequence:

```
#ifdef HELTEC_V3
  if (LT.begin(NSS, NRESET, RFBUSY, DIO1, LORA_DEVICE)) {
    Serial.println(F("LoRa Device Heltec V3 found"));
    display2 = "LoRa Device HV3 found";
    displayData();
    led_Flash(1, 125);
  } else {
    Serial.println(F("Device error"));
    while (1) {
      Serial.println(F("No device responding"));
      display2 = "No LoRa Device found";
      displayData();
      led_Flash(50, 50);  // long fast speed flash indicates LoRa device error
    }
  }
  LT.setDIO3AsTCXOCtrl(TCXO_CTRL_1_8V);
#endif
```

With this additional code line, the receiver code runs like expected.
Greetings
Michael

StuartsProjects, 2025-01-22:
I would be cautious about using setDIO3AsTCXOCtrl() out of the sequence of setup steps used in the setupLoRa() function. 

Whilst a single setDIO3AsTCXOCtrl() might appear to work, there could be circumstances, particular bandwidths, spreading factors, packet lengths etc etc, where problems develop. 

So far the sequence in setupLoRa() has been used for some years with no known issues. 

Thanks for the confirmation that Heltec appear to be using an unusual TCXO voltage.

closed 2025-01-22 by AndroidCrypto