StuartsProjects/SX12XX-LoRa #106: Cant get CAD interrupt working on SX1280

ChSt15, 2025-04-13:
Im working on a datalink system and to make it more reliable, I need to use the channel activity detected function of the SX1280. I enable all IRQ interrupts and constantly check if the DIO1 pin goes high, retrieve the IRQ status and check if IRQ_CAD_ACTIVITY_DETECTED is flagged, buts its never flagged. Everything else seems to work (RX_DONE, TX_DONE etc.). I noticed there are settings for the CAD symbol length, but no where to set it.

Is the CAD function of the SX1280 implemented or not?

StuartsProjects, 2025-04-13:
There are a number of SX1280 functions that have not been used in the working examples. 

I have not tried the CAD detect function on the SX1280 myself. 

The datasheet states that you need to use the SetCadParams to setup the CAD detect then enable the interrupt. 

Check the SX128XLT.CPP file for the library function writeCommand() which will allow you to setup the CAD. There are plenty of examples in  SX128XLT.CPP on how to use writeCommand(). 

There are also examples of how to write commands to the SX1280, not using library functions, in the example 2_Register_Test.ino

ChSt15, 2025-04-21:
Is there a way to also find out is the receiver is currently receiving a packet? Currently I set the receiver to always be in receive mode and then switch to transmitting whenever i need to send something. Im getting the issue that when its in the middle of receiving a packet, I switch to tx mode to send a packet and that means loosing the packet. Is it possible to detect an incoming packing using the registers or IRQ interrupts? I would use this is delay the sending until after the packet has been received.

ChSt15, 2025-04-21:
I found out in the Datasheet it says that SyncWordValid is not LoRa compatable. Preamble detected is the correct interrupt that can be used for this. 

I should add for possibly others. To use the CAD, you must do these commands:

`
    setMode(MODE_STDBY_RC);
    setDioIrqParams(IRQ_RADIO_ALL, (IRQ_CAD_ACTIVITY_DETECTED + IRQ_CAD_DONE), 0, 0); 
    rxEnable();

    writeCommand(RADIO_SET_CADPARAMS, [cadLength], 1);

    writeCommand(RADIO_SET_CAD, NULL, 0);
`
cadLength is the number of symbols to use for CAD. These are given in the library definitions.