StuartsProjects/SX12XX-LoRa #107: Project 240_StuartCAM_ESP32CAM no pictures

davefes, 2025-08-07:
Have been trying to get 240_StuartCAM_ESP32CAM to take pictures for a few
days now.  I can capture a picture (frame buffer) using the tutorial on 
RandomNerdTutorials:
https://randomnerdtutorials.com/esp32-cam-take-photo-save-microsd-card/
but not using project 240.  I have compared all the settings and can not
see what is different.

I am using a AI Thinker ESP32-CAM and Arduino IDE 2.3.6 on a Ubuntu machine.
I have gone through all the suggestions that AI have thrown-up.  Tested
on three ESP32-CAMs and two cameras.

This is not a library issue, but was hoping that someone may have tried
Project 240 recently.

Thank you.

StuartsProjects, 2025-08-07:
The library code has been there for some years, no errors reported. 

Can you provide more information please.

Is the serial monitor reporting an error, if so what is it ?

What version of the Expressif ESP32 core are you using ?

davefes, 2025-08-07:
On serial monitor
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4980
load:0x40078000,len:16612
load:0x40080400,len:3480
entry 0x400805b4

Awake !
Bootcount 1
Sleepcount 0
Payload CRC enabled
Initialising the camera module 
PSRAM found
got to takePhotoSend()
just before camera get
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (cam_task) 
Core  0 register dump:
PC      : 0x4008b2d6  PS      : 0x00060a36  A0      : 0x8008f154  A1      : 0x3ffb2550  
A2      : 0x3ffb0300  A3      : 0xffffffff  A4      : 0xb33fffff  A5      : 0x3f40ef18  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x00060a23  A9      : 0x3ffb2550  
A10     : 0x00000003  A11     : 0x00060a23  A12     : 0x00060a23  A13     : 0x0000001f  
A14     : 0x3ffb2ab0  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40087749  LEND    : 0x40087759  LCOUNT  : 0xfffffffe  

Backtrace: 0x4008b2d3:0x3ffb2550 0x4008f151:0x3ffb2580 0x4008f186:0x3ffb25a0 0x4008f2fc:0x3ffb25c0 0x4008314f:0x3ffb25e0 0x40083169:0x3ffb2610 0x40082e5d:0x3ffb2630 0x4008b7c0:0x3ffb2650 0x4008ab11:0x3ffb2670 0x4008ad54:0x3ffb2690 0x40083fa6:0x3ffb26b0 0x40083fee:0x3ffb26d0 0x40084151:0x3ffb2700 0x400e0fad:0x3ffb2720 0x400e3261:0x3ffb2750 0x4000bd83:0x3ffb2770 0x4000182a:0x3ffb2790 0x400e2cfd:0x3ffb27b0 0x400e3261:0x3ffb27d0 0x40087516:0x3ffb27f0 0x400869b9:0x3ffb2810 0x40086a0a:0x3ffb2830 0x40086f09:0x3ffb2850 0x400ffde7:0x3ffb2880 0x400ff7fe:0x3ffb28a0 0x400f6525:0x3ffb2bc0 0x40090395:0x3ffb2bf0 0x400e66e4:0x3ffb2c50 0x4008b1d1:0x3ffb2ca0

ELF file SHA256: ac13123a2
```
Added some debugging messages.

Expressif ESP32 core?  Tried to do:
`esptool.py  --chip esp32 chip_id`
and no luck.  
The can is marked ESP-32S

I will come back with more information.

davefes, 2025-08-07:
I can read the chip_id on ESP32-WROOM boards, but not on the ESP32-CAM.  If you mean which ESP-IDF version, I have no idea.  I would guess it is part of the IDE.

davefes, 2025-08-07:
Maybe, another issue I am having might shine some light ... on all scripts I can not get any SD cards to work.  Even on RandomNerdsTutorial scripts.  This does not worry me as I hope to stream the data on the receiver via WiFi.

davefes, 2025-08-07:
AI says ... Arduino IDE 2.3.6 uses the latest stable version of the [ESP32 core for Arduino], which is version 3.0.0.

StuartsProjects, 2025-08-07:
You need to check and report which version of the ESP32 core you are actually using, there have been about 15 versions since 3.0.0 was released.

davefes, 2025-08-07:
It has been a few years since I did any Arduino stuff, but your LoRa reliable packet look too interesting.  Looks like it was a beginner's mistake.  I changed the flash mode from what the IDE had as a default (QIO) to DIO.  240 sent a picture to 241.

I think the major-hurdle has been overcome.

Thanks for the support.

closed 2025-08-07 by davefes