StuartsProjects/SX12XX-LoRa #108: Projects 240/241, CRCs do not match

davefes, 2025-08-13:
I can get 'pictures" across on a 240/241 link.  I get lengths match,
but CRCs do not match.

`2 Source array CRC is 0x0
3 Source array CRC is 0x0`

I am guessing that the `2 Source arary CRC` did not make it to the RX.

As I have been having troubles using SD cards on the ESP32-CAM and
the fact that I do not need the files saved on the endpoints, I
have stripped out all the SD-related code from 240 and 241.
I only left
`bitSet(ARDTflags, ARNoFileSave);  // no SD card being used`
in 240, in case it was "telling" the TX/RX something important.  I
don't see anything in 241 that could "break" if there is no SD.

Could you suggest where I should focus my attention to sort the
CRC issue out?

On my Micropython ESP32-CAM installations the picture files have the
correct finish marker for JPEG on them.  I don't see this marker
in Serial Monitor when I print out the contents of PSRAM.

Do the start and finish markers have to be added manually?

I intend on sending the contents of PSRAM to the WiFi system.

closed 2025-08-14 by davefes

StuartsProjects, 2025-08-14:
So try the provided example and when its working, try taking the SD stuff out and see what happens.

davefes, 2025-08-14:
Getting these two projects working with SD cards would be my preferred option.